tomjon/piSpecMon #24: Soundcard Audio Out

jennyb, 2016-09-19:
When placing a PSM at a location, the customer would like to listen to the receiver audio to verify that they are listening to the correct station. Unplugging the speaker connector from the radio and turning the radio volume up can cause problems when the customer incorrectly returns the PSM to normal use. 
Instead there should be audio output from the soundcard ( Pi,external,DAB ) that the operator can listen to with headphones. That may be set on/off/volume with the UI

tomjon, 2016-09-21:
[image] #24</a>

tomjon, 2017-08-30:
The external sound card has audio out which has monkeyboard on left channel and receiver on right channel. The output volume can be controlled by amixer so could provide a UI control for this on the Audio panel. Currently there is no way to access the audio out socket externally, however.